NeonHorizon/lipopi #2: 4.2Volts on 3.3Volts GPIO

mvdswaluw, 2015-12-18:
The low power pin on the powerboost 500c will go as high as the battery-voltage (4.2 Volts). You connect this to the 3.3Volt GPIO directly. In order to protect your GPIO you should drop the voltage and limit the input current. I've done this by using a 1n4148 and a resistor in series.

bmishra4, 2016-01-12:
Hello mvdswaluw,
If possible, kindly share your workaround of this. I don't have any knowledge about electronics, just got raspberry pi and trying to implement this. A picture or a diagram of connections will be a great help.

Thanks
Brij

NeonHorizon, 2016-01-30:
For some reason I've not been getting notifications, so sorry for the delay in response....

The low battery pin on the PowerBoost only goes high when the battery is flat, IE when its at 3.2v. Therefore you do not need any diodes or resistors because it never goes above 3.3v. Its impossible for the low power pin to reach 4.2 volts as you describe.

mvdswaluw, 2016-01-30:
Not true. On the powerboost 500c the pin is high untill the battery is below 3.2v than it drops to 0v.

NeonHorizon, 2016-01-30:
Ooo.. interesting, you might be right. Let me double check this. My notes say it goes high when the battery is flat but Adafruit says its the other way.

mvdswaluw, 2016-01-30:
Question: Would a circuit be able to go high when the battery is low? ;-)

NeonHorizon, 2016-01-30:
Yes. The battery is low at 3.2v which is enough voltage to go high ;)
If it the battery was so low there wasn't enough voltage to go high the Powerboost and PI would be powered off anyway so it wouldn't matter ;)

But anyway I think you are correct, I am just testing to make doubly sure now then I will update the text in the document. I'll also check the Powerboost circuit to see if there is any resistors between the battery and the pin when its high. I have to say I've been using this setup for a while now so the 4.2v seems to be a non issue, but we should do it correctly like you say.

NeonHorizon, 2016-01-30:
OK I've corrected the documentation I'm going to look at this Adafruit circuit diagram.

NeonHorizon, 2016-01-30:
Hmmm... I wish Adafruits diagrams were a little better.
Looking at this it doesn't state exactly where the LBO output comes from:
https://learn.adafruit.com/assets/17555
I'm guessing it can't be the output on the transistor which is connected to the LED as that would result in the line going high when the battery is flat as thats when the LED lights up (thats probably why I got confused). I'm guessing it must be the TPS6109, I'll check the datasheet for that.

mvdswaluw, 2016-01-30:
It seems the step up chip has a open collector output. They use the internal resistor of the mmun2133 (4k7) as pullup. So there is a resistor there, but not physical. Open circuit is would be as high as vbat, but voltage will drop when loaded.

NeonHorizon, 2016-01-30:
Yeah that's the same conclusion I was just writing.
If I understand the MMUN datasheet correctly though its not 47k not 4k7 and the current would have to go through two of them totalling 94k. 
http://www.weitron.com.tw/PDF/MMUN2111.pdf

I'm pretty sure the risk of the overvolt at that impedance is non existent which is why its not blown up my Pi's yet.

NeonHorizon, 2016-01-30:
Oh hold on its a 2133 not the 2113, so its a 4k7 plus a 47k which gives us 51k7

craic, 2016-03-31:
The LBO pin on the PowerBoost 500C mirrors the battery voltage until the latter hits 3.2 and then it drops.

I set up an ADC so I could measure the decay in battery voltage over time. The peak voltage on the LBO pin is 4.2V when the battery is fully charged so I used a 22K and 100K resistor to create a voltage divider that drops the max voltage down to 3.3V for the ADC.

One wrinkle in the experiment - for some reason the measured voltage on the LBO pin via the ADC is lower than it really is - and the LBO red LED is ON in this configuration... so there is some current path that I don't understand - but measuring the LBO voltage directly with a multimeter (when not connected to the ADC) shows it to be the same as the battery.

Here is the plot - 
[image]

The battery continues supplying power to the end of the plot at which point the Pi shuts down (I was not using the low battery pin to shut it down in this experiment).

NeonHorizon, 2016-03-31:
Looking at the specifications for the TPS61090, the LBO pin is an open drain. That means it pulls the LBO line low (to ground) when the battery is flat but its open circuit the rest of the time, IE: it can't actually source current it doesn't generate a voltage.

This means the voltage you are measuring is not from the chips LBO pin but from R2 +R1 inside Q1 (the LED transistor) which are 47k + 4k7. So basically when the battery is not low and the chips drain is open circuit your voltage divider is essentially sitting between ground and these 51.7k of resistors inside Q1 to the battery voltage. That's why you get the offset.

The reason the LED is lighting up is you are sinking enough current through your voltage divider to trigger the transistor and hence light up the LED. Basically you are pulling the output from LBO on the chip low(er) even though you are using a total of 122k.

This is basically why saying the LBO output will damage the Pi isn't true. The LBO pin may be at the battery voltage but the impedance is so high even the slightest current draw will pull it low. And my understanding is you wont be able to damage the GPIO on the Pi unless there's actually some current there to do the damage.

craic, 2016-04-01:
Yes that does indeed make sense - thanks for the explanation !

Jud, 2016-05-16:
@craic @NeonHorizon looking at this thread and the 'on/off version', do we actually need the diodes shown on the schematic?

craic, 2016-05-16:
The two diodes on the 'Power Down' path serve to drop the voltage from the
Battery by 1.4V (0.7V per diode x 2) to make it in safe range for the GPIO
pin which can tolerate 3.3V max. You could do this with a couple of
resistors to create a voltage divider.

The one on the 'Power Up' path serves to isolate that from the 'Power Down'
side. When the Pi is booted there will be 3.3V on GPIO 14 - that is what
keeps the PowerBoost Enable Pin high and, therefore, operating.

Without that diode GPIO 18 would see 3.3V and would immediately trigger a
shutdown of the Pi.

On a separate note, my breadboard diagram shows the switch connected to the
Vs pin of the 1000C - this should be the Bat pin - the Vs pin can go up to
almost 5V - I'll change the diagram as soon as I get the chance.

--Rob

closed 2016-06-14 by NeonHorizon