NeonHorizon/lipopi #8: Not latching on

Iridium69, 2016-06-07:
Hi, I have a new zero with camera module and have set up the PowerBoost1000c as per the on off version.
Everything works except the Pi will only stay on when holding the button.  Even holding the power button in until login but the PB powers off as soon as the button is released.

When powering from Pi USB power off via button works fine with the script running.

Any help massively apreciated.

Mark

craic, 2016-06-07:
Did you turn off the Serial option in raspi-config ? If you did not then it
will power off automatically

NOTE - Before doing anything, be sure to run sudo raspi-config and under
"Advanced Options" select "Serial" followed by "No". This prevents the Pi
using GPIO 14 for the console (which would shut off the power).

Iridium69, 2016-06-07:
Thanks for getting back to me so quickly!
Yep, checked the raspi-config and serial is definitely off.

craic, 2016-06-07:
OK - next thing to check is the circuit - check the power up side is connected to GPIO14 (4th pin down from the SIM card end on the **outer** row of pins) - it has to be GPIO14

A couple of other questions that might be helpful to know:
Are you running the python script ? (not needed to test the power up side of things) - and are you running it under Systemd ?

If you have a photo of the hardware that would be great

Iridium69, 2016-06-07:
I hope that you can see from these pictures.
Please forgive me I am a novice at component level. It's a tatty way to wire it up but I think it follows the circuit diagram.

craic, 2016-06-07:
Sorry, don't see a photo...

Iridium69, 2016-06-07:
Oops, forgot to mention I'm currently using the python script in terminal.  Haven't got as far as making it run under systemd

The shutdown part of the script works fine when the script is running. I just press the button and she gracefully shuts down.  I managed to do this by plugging in USB power to the Pi.

Thanks,
Mark

Iridium69, 2016-06-07:
I'll try and only attach one this time...

craic, 2016-06-07:
OK - don't try using systemd until _everything_ is working smoothly

The power-up part works independently of the script (it has to as the Pi hasn't fully booted yet !) and you could just disconnect the power-down path to help isolate the issue

Iridium69, 2016-06-07:
[image]

Iridium69, 2016-06-07:
[image]
One last pic before I hit the hay for the night.
Thanks for helping.
More tomorrow, fresh eyes and all ;)
For some reason Pin 8 on the Pi isn't keeping en on the PowerBoost alive.

craic, 2016-06-07:
Got the photo - two things come to mind:

#1 That's a lovely work mat
#2 There is no battery attached in the photo - you are testing this with a lipo battery plugged in, correct ? You must be, or nothing would happen... and the lipo is not discharged ?

The visible parts of the circuit look correct 

I would always recommend using a breadboard first - get it all working first before solder and shrink-wrap

update - just saw your second photo without the tubing - I'll digest that and get back to you

--Rob

NeonHorizon, 2016-06-08:
Yeah wheres the battery :)
If it goes off straight away when you let go this means either then enable line isn't high on the Adafruit or the battery is flat.

craic, 2016-06-08:
Your circuit looks OK from the photos - so here are the steps that I would take to troubleshoot the problem - they may seem simplistic but whenever I have this sort of problem it is usually something simple...

#1 Check your LiPo battery is a 3.7V model and is charged - and is big enough to handle a Pi (at least 500mAh and prefereably 1000 mAh or more)
#2 Make sure there is no chance of a short on the back side of your PowerBoost - your soldering is a little, erm, heavy - so check for shorts between the PowerBoost pins
#3 make sure there is no short between the back of the PowerBoost and the Pi
#4 Boot the Pi on USB and test the voltage between GPIO14 and ground - should be 3.3V
#5 unsolder a connection to break the Power Down path - so you just have the Power Up path - see if the problem persists
#6 unsolder all the connections - check that the PowerBoost is working as expected (i.e. charges the battery) - check that the Pi works as expected
#7 rebuild the circuit with new diodes, resistors and switch - and a breadboard :^)
#8 Watch the Pi boot messages on the console and try and catch any errors that fly by (and check the various system logs)

The fact that it works fine on USB is puzzling...

Iridium69, 2016-06-08:
Thanks, that's very helpful.
LiPo is reading 4.2v so that's ok
LiPo is 500mAh 3.7v
Insulation taping the Pi in a sec ;)

Using a 2.4A USB power adapter and measuring only 0.57 v between GPIO 14 an ground!

Aha!

Now I need to figure out why....

Back to the breadboard then!

Thanks for your help man, much apreciated!

Iridium69, 2016-06-20:
Hi,
I went back to breadboard today, fresh set of components, same result. It will still not stay on.

Things to note:
I'm measuring the resistors I'm using at 9.5k and 110k, not quite per instructions but close enough?
Diodes are N4001 and new.
i am getting 1.5 - 2v on GPIO 14 when charging the battery.
0.5 v with just the battery plugged in.

When booting from the battery or USB power to PB1000 GPIO 14 blips but stays low (drops to 0v pretty much straight away).  
I'm measuring from GPIO 14 to ground pin on Pi.
However when powering from USB to the Pi GPIO 14 goes to 2v.

As soon as I can I will get another zero and make sure mine isn't faulty but they are hard to get at the moment!

Thanks,
Mark

Iridium69, 2016-06-20:
[image]

[image]

Much neater, hope you can see how it's wired...

Iridium69, 2016-06-20:
[image]
The first photo looked wrong but you can see better here how it's connected.

NeonHorizon, 2016-06-20:
This could be a software thing.
If you disconnect everything from the Pi and just boot it from a normal USB power supply is GPIO14 high?

Iridium69, 2016-06-20:
Yes, 2.5v with just the Pi alone.

NeonHorizon, 2016-06-20:
That seems a little low. Do you definitely have the serial console disabled?

Iridium69, 2016-06-24:
Just got around to checking and yes serial is definitely disabled.

I have a new Pi Zero to try will report back as soon as it is hooked up.

Iridium69, 2016-06-24:
Aha! 
The newer Zero with the camera port is only giving 1.5v on GPIO 14 whereas the original Zero is giving 3.9v!

Maybe that's why it isn't latching on...

NeonHorizon, 2016-06-24:
Well the old one shouldn't be 1.5v and the new one shouldn't be 3.9v so you're not having much luck there :)

But presuming that the new one is closer to 3.3v where it should be and you just have a dodgy meter, I wonder what the issue with your newer zero is. I've not got one myself but it sounds like its either faulty, sending data down GPIO14, or the newer zero has the same UART issue as the Raspberry Pi 3.

Iridium69, 2016-06-24:
Eureka!
It's working!

The difference in voltage is a software thing, I switched from the latest raspbian lite to an older version of raspbian on a different card and that sorted out the voltage on GPIO 14.
Next I noticed that there were dips in the voltage on GPIO 14 as the Pi booted past the recovery splash screen.  I have to hold the power button until the boot text is rolling and then it stays latched on.

Ah that feeling you get when you figure out some technical gremlin like this is awesome!!

Now I can get the project finished and mounted in the rocket! 

Thanks for your help!!!

Mark

NeonHorizon, 2016-06-24:
Well done Mark.
Interesting about the Raspbian thing, I wonder if anyone else will report it.

generalectric, 2016-07-06:
I got the same exact problem, been stuck for days. Managed to fry my first zero so I got another and have the same prob. camera rev v1.3. What version of raspian are you using?

NeonHorizon, 2016-07-06:
Its interesting that the version of Raspbian should have an effect, it shouldn't do since the pullup on GPIO14 is hardware inside the BCM not software.

Iridium69, 2016-07-06:
Ill check when I get home tonight, I will check the version of Raspbian it works with and recheck that the latest version of Raspbian Lite has an issue.
Standby...

generalectric, 2016-07-07:
BLAMO! Got it... GPIO14 was low. Just add the following to lipopi.py.
`GPIO.setup(14, GPIO.OUT)
GPIO.output(14, GPIO.HIGH)`
Also had to place a 1000uf capacitor parallel with the 100k resistor as the 100uf I had wasnt enough to bridge the entire boot, Bad news is full power off takes a couple minutes due to oversized capacitor...

NeonHorizon, 2016-07-07:
Interesting hack, you really shouldn't have to do that though, GPIO14 should go live at boot. What we need to figure out is if the issue is a new version of Raspbian or the physical PiZero with the camera slot.

Iridium69, 2016-07-07:
Sorry for not getting back to you guys yesterday, I will try again tonight, see if we can nail this one down.

Iridium69, 2016-07-08:
Sorry it's taken so long but the results from uname -a 
Linux raspberrypi 4.4.13+ #894 Mon June 13 12:43:26 BST 2016 armv6l GNU/Linux

rsend, 2016-07-26:
Hi, has this been sorted? I just got a 1.3 and am about to build myself another of these (works beautifully on the previous versions).

Other question, I CANNOT move GPIO 14, but I can move the other two correct?

Iridium69, 2016-07-27:
Hi Rafael, mine is working fine with the PB1000C and the new Pi Zero.  My issue turned out to be a software one.
Not sure why the raspbian lite I first tried didn't latch, I haven't figured out why as the version I am now using works fine.

As far as I know GPIO14 is the only pin which will go high on power on so you are correct, this is the one you cannot reassign.  The other two you can no problem.

I should have my first project prototype working this week now I have sorted my solder station out properly 😳

NeonHorizon, 2016-07-27:
Just confirm what Iridium69 said, yes you can move any of the GPIO's except 14.
If it works on the Zero 1.3 with the latest Raspbian please let us know then I think we can close this bug?

rsend, 2016-08-12:
Hi again-
I've just finished building the circuit and I'm running into the same problem. What did the solution end up being? I needed to update all the software to get some other things running, so I hope the downgrading was not the only fix.
Any word on this?
Thanks,

Rafael

generalectric, 2016-08-16:
Sorry I've been trying to wait for other feedback as I'd hate to give bad or misleading information. What works for me across three different builds may or may not work for you. In my case I only need a capacitor if I'm not ok with holding the power button 10-15 seconds until my startup script sets GPIO14 high. Once the script sets GPIO14 high it does in fact latch and stay on.

rsend, 2016-08-17:
Ok. 
Where in the script should I put the actual pin setting command? I tried it in the setup section along with the pin setup command and it went low (and shut down) after a few seconds.

R

tjlage, 2016-08-17:
Hi guys, I can confirm the same issue happening to me. I'm using a RPi Zero v1.3 and RetroPie 3.8.1. I'm not sure which version of Raspbian, RetroPie 3.8.1 is built off of at the moment but I'll be able get that info tomorrow.

NeonHorizon, 2016-08-17:
Do we know if this is an issue with the new Raspbian or the Zero 1.3 yet?

Iridium69, 2016-08-17:
Yes, the issue was there on the origional zero and on the new one with the exact same setup.  When I switched the OS the issue went away so I suspect it not to be a hardware issue but a software one.

NeonHorizon, 2016-08-17:
If its software its likely to be something using trying to use the UART. I wonder if the new software is trying to see if the Pi3 Bluetooth module is present? There might be a kernel module we can disable or something?

tjlage, 2016-08-17:
I used the uname -a command on my setup and it is the same as the OP.

Linux retropie 4.4.13+ #894 Mon Jun 13 12:43:26 BST 2016 armv61 GNU/Linux.

generalectric, 2016-08-17:
Add this line to /etc/rc.local
`python /home/pi/latch.py &`

latch.py:
`#!/usr/bin/env python`
`import RPi.GPIO as GPIO`
`GPIO.setmode(GPIO.BCM)`
`GPIO.setup(14, GPIO.OUT)`
`GPIO.output(14, GPIO.HIGH)`

Hold the power button roughly twelve seconds until latch.py is executed in the boot process ensuring GPIO14 is in fact high at which point you should be able to release the power button and maintain your pi's power state.

Please post your results!

generalectric, 2016-08-17:
Tj you did have to set GPIO14 high correct? I mean in addition to the capacitor?

tjlage, 2016-08-17:
I'll test out that latch py script when I get a chance.

I had to re-enable the serial communication, then put the 470uF capacitor across the 100K resistor. It latched immediately after I pressed the switch.

generalectric, 2016-08-17:
I just retested mine. If I comment out latch.py from rc.local GPIO14 doesn't go high and my pi loses power right as it finishes booting. Tested on three boots.... When I removed comment from latch.py line in rc.local it booted up and stayed powered on.

Linux localhost 4.4.16+ #899 Thu Jul 28 12:36:19 BST 2016 armv6l GNU/Linux

rsend, 2016-08-17:
I added your latch script and it works fine now, if I hold the button past boot.
Thanks!

R

closed 2017-02-07 by NeonHorizon