NeonHorizon/lipopi #12: can't see the button in the photo

The-Alchemist, 2016-08-04:
Great works, guys!

Unfortunately, I'm a little blind, I can't seem to find the button in the wonderful photo:

[image]

mindforger, 2016-08-05:
it's the one on the far right side, above the USB connector, facing to the side, wrapped in tape or shrinkwrap

a little bit hidden by the black SD card sticking out beneath

NeonHorizon, 2016-08-05:
Yep thats the one :)
I put it there to make the whole thing as slim as possible and also have it accessible next to the USB connector on the edge.

closed 2016-08-05 by NeonHorizon

The-Alchemist, 2016-08-05:
Oh, it was wrapped in tape.  That's why I had trouble seeing it.

Thanks, guys!