NeonHorizon/lipopi #16: Shutting straight down

urban2015, 2017-02-15:
I have followed the writing diagram and i have followed the installation of the software but my pi zero keeps shutting down when i let go of the power button. I do press and hold the button for at least 2 seconds.

I have the Pi Zero with the 500 charger board.

Any ideas ?

NeonHorizon, 2017-02-15:
When you power it up, what voltage is on GPIO14?

urban2015, 2017-02-15:
When i hold the power button down and measure the voltage on the that pin it is 3.4

craic, 2017-02-15:
Have you Run sudo raspi-config and under "Advanced Options" select "Serial"
followed by No. This prevents the Pi using GPIO 14 for the console (which
would shut off the power). That step is essential (except on the Pi3!)

Also you might want to add a 0.1 uF capacitors between GPIO26 and ground

Take a look at https://github.com/craic/pi_power which I based on the
LiPoPi circuit - look for where I talk about the capacitors - I had a
couple of issues when working on a breadboard with relatively long leads,
probably due to interference. Adding the capacitor helped there.

--Rob

urban2015, 2017-02-15:
I have run the command to disable the serial. I will look at that link you have sent as well.

NeonHorizon, 2017-02-15:
Silly question but its worth checking, is it 3.4v on the 500 board enable pin as well.

urban2015, 2017-02-15:
Its 3.43 on the enable pin

NeonHorizon, 2017-02-15:
Thats odd... I guess it could have data on it which might be causing the problem.
Before we go further though is your battery OK?
What voltage is on the batt pin?

urban2015, 2017-02-15:
There is 2.2v on the batt pin on the charger.

urban2015, 2017-02-15:
I am just looking at the advanced options in the config and the serial option is missing. I am now running an update on it.

craic, 2017-02-15:
2.2V ? That doesn't sound right - You are using a 3.7V LiPo battery right ?

I'm not sure how you get down to 2.2... must be completely drained

If it is not a 3.7V LiPo then it won't work

If it is, then charge it up fully and see how that works

The circuit should be working, but even a Pi Zero sucks quite a bit of
current - you might want to upgrade to PowerBoost 1000C

--Rob

urban2015, 2017-02-15:
Here is my set up

[image]

craic, 2017-02-15:
That jpg didn't come through - perhaps you sent the email before it had
fully uploaded - try it again

urban2015, 2017-02-15:
Working now ?

urban2015, 2017-02-15:
[image]

urban2015, 2017-02-15:
I will leave the battery on the charge for a while but without the wires in place only connecting it to from the pi to the charger with the + - the pi will run fine.

craic, 2017-02-15:
If you have a 0.1uF capacitor, try putting it between GPIO18 and ground (I gave you the wrong pin number earlier). Your wires are pretty long and I've run into interference issues with that in other projects - worth a try...

NeonHorizon, 2017-02-15:
2.2V on that LiPo pack is really really low. I wouldn't leave that unattended when you are charging it. Also its definitely the reason why your LiPoPi isn't working. At 2.2v you might not even be able to charge it.

urban2015, 2017-02-15:
I will check the battery voltage without it connected to the charger 500.

urban2015, 2017-02-15:
Its 4.0v when its not connected to the charger 500.

urban2015, 2017-02-15:
If i plug the battery in now it comes straight on !!! werid

urban2015, 2017-02-15:
OK i have it all connected and at the battery connecter at the back it measures 4volts

urban2015, 2017-02-15:
I have got a 0.1 capactor when you say pin 18 and ground do you mean ground on the pi zero pin out and pin 18 on the pi zero ?

craic, 2017-02-15:
correct - GPIO18 and Pi Ground

urban2015, 2017-02-15:
ive added that and no didnt make a difference.

[image]
[image]

urban2015, 2017-02-18:
How to remove the lipopi.service? so i can try again. I am getting failed to load the lipopi.service .

I am trying the raspberian lite version os now

urban2015, 2017-02-20:
When I install the lipopi.service script should the pi keep shutting down without the bread board connected ?

urban2015, 2017-02-20:
Now i install the this service my pi keeps shutting down without the bread board connected

NeonHorizon, 2017-02-20:
Yes thats correct.

urban2015, 2017-02-20:
ok now when i connect my bread board and boot up the pi by holding the power button the pi boots up and i can see that the lipopi.service is runnning ok. I keep my finger on the power button till the login comes up and i let got of the power button and the power goes from the pi.

Can you help any more as i have now re done the bread board and measured the resistors to make sure i have 2x 10ks and 100k.

I am running rhe raspi os lite. With the serial disabled.

I would love to get this working.

NeonHorizon, 2017-02-20:
As far as I can tell from looking at your photo its wired up correctly. As long as that breadboard doesn't have any dodgy connections. Does it stay running if you try it with the service disabled?

urban2015, 2017-02-20:
Here is new photos.

[image]
[image]
[image]
[image]

urban2015, 2017-02-20:
ok i have disabled the lipopi.service and without the breadboard connected the pi stays on. With the bread board connected the pi turns off when i let go of the button.

urban2015, 2017-02-20:
I have noticed though when i do the apt update and go to the config page the serial is no longer in the advanced section. But the serial is still disabled.

NeonHorizon, 2017-02-20:
With the service disabled and the board disconnected you are definitely seeing 3v3 on GPIO 14 right? Because what you are describing sounds like GPIO14 is not going high on power on.

urban2015, 2017-02-20:
OK with the bread board disconnected and running from the battery the GPio 14 is 0 all the time.

NeonHorizon, 2017-02-20:
Something is wrong with GPIO14 then. It should go high whenever the Pi is running even when its booting.

urban2015, 2017-02-20:
ok if i enable the serial from the config page in the inconectable options i get the 3.3 voltage back

NeonHorizon, 2017-02-20:
You shouldn't need serial enabled but if you do use the data as the high signal you might need the capacitor mod, try it without and see.

urban2015, 2017-02-20:
What do you mean by use the data as the high signal ?

urban2015, 2017-02-20:
So without the script running the pi will stay on if i let go of the power button when i get the login screen. If i enable the script the pi continues to go off straight way. If i run the lipopi.py script when the service is disabled then i can shut the pi down with the power button.

NeonHorizon, 2017-02-20:
If you turn on the serial port then there will be data on it. This can be used to simulate a high signal.

I'll let Rob get back to you about the scripts.

urban2015, 2017-02-20:
I am wondering if the latest version of the config is different and creating problems. I will try later with the old version of the cofig and see if that changes anything. Thanks fro your help so far.

urban2015, 2017-02-20:
OK i think i have it working :)  The config file on the serial had 2 settings first one turn off data to the serial and second one actaully turn of the serial hardware. I had turned both off duh !

OK it seams to work nicely now.

NeonHorizon, 2017-02-20:
Oh thats really interesting!
Which settings did you use, we should clarify on the main pages.

tommcintyre, 2017-02-25:
I can confirm - looks like there have been some changes to raspi-config menu layout. The settings are now located under "5 - Interfacing Options" / "P6 Serial". It is then necessary to answer as follows:
"Would you like a login shell to be accessible over serial?" - No
"Would you like the serial port hardware to be enabled?" - Yes

(This is on a Raspberry Pi Zero. I also added the 0.1uF capacitor between GPIO 18 and Pi Ground for good measure, although it didn't seem to be necessary. :) )

NeonHorizon, 2017-02-26:
Thanks Tom, I've added your details to the readme and put your name in the credits.

Does anyone know if this solves the issue with the RaspberryPi 3?

Rob I'll leave you to update the pi_power documentation

craic, 2017-03-03:
I see the changes to raspi-config... I wish they included a version number in that... 

I'll try and test this out with my circuit sometime in the next week - can't get to it sooner, unfortunately

closed 2017-08-30 by NeonHorizon