NeonHorizon/lipopi #17: Need to use serial to connect pc to Pi

PunyViking, 2017-03-08:
Do I need to use the UART pins on the raspberry for the LiPoPi to work? I need to access the serial port on my pi zero to communicate between Pi and pc, can I use some other gpio.

NeonHorizon, 2017-03-08:
The GPIO 14 (UART TXD) is the only pin which goes high at power on and low at power off. If you check the issues, others have used different pins but they have to write a script to make the GPIO high and hold the power button down until the Pi is booted.

closed 2017-03-08 by NeonHorizon

jsinkers, 2019-07-17:
There are a number of pins that boot to HIGH.  e.g. for rpi0 [BCM2835 Arm Peripherals](https://www.raspberrypi.org/app/uploads/2012/02/BCM2835-ARM-Peripherals.pdf) 6.2 Alternative function assignments indicates the pins that go high at power on are:
    * GPIO/BCM: 0, 1, 2, 3, 4, 5, 6, 7, 8, 34, 35, 36

So these pins should be able to be substituted for UART.  Obviously some of these pins have other functions that you may not want to interfere with so it's up to you to pick which one works.  
I'm not sure if this is the same for other rpi's - you will have to check the documentation.

I am using pin 5 at the moment and I don't have to hold the power button down for it to start up.

But there is another change necessary to ensure that the pin goes low on power down, here I used the power-off dtoverlay.
    ```bash
    $ sudo nano /boot/config.txt`
    ```    
    Add the following line:
    `dtoverlay=gpio-poweroff,gpiopin=5,active_low="y"`
    Replace "5" with whichever `power_disable_pin` you are using.

This worked for me - hope it helps.

NeonHorizon, 2019-07-17:
There are other pins which boot to high but they don't go high straight away and stay high until the end of shutdown as far as I know?

jsinkers, 2019-07-18:
I should have mentioned this is using the RC power up/power down circuit. The RC response gives enough time for the pin to go high. The dtoverlay ensures the pin goes low to disable power at the end of shutdown. 
People have used scopes to examine this behaviour here: https://www.raspberrypi.org/forums/viewtopic.php?t=201483

karlahaiat, 2019-07-22:
Excuse me for the beginner question, but what is the RC power up/power down circuit?

jsinkers, 2019-07-23:
@karlahaiat See [Power-up/power down](https://github.com/NeonHorizon/lipopi/blob/master/README.power_up_power_down.md) here.

karlahaiat, 2019-07-29:
@jsinkers I was able to setup the shutdown with another pin, but are you also able to power it on with a different pin (other than pin 14?)

karlahaiat, 2019-07-29:
@jsinkers I was able to get this to shutdown with other pins, but am not able to get it to turn on again, did you do anything special to make this work?

jsinkers, 2019-07-31:
(Re: @jsinkers <https://github.com/jsinkers> I was able to get this to
shutdown with other pins, but am not able to get it to turn on again, did
you do anything special to make this work?)

@karlahaiat did you select one of the pins that goes high when the rpi
starts up?  See for example my response here
https://github.com/NeonHorizon/lipopi/issues/17#issuecomment-512169331

What pin are you using?  What behaviour are you seeing when you press the
button? Are you using the RC circuit?  Please add some more info