NeonHorizon/lipopi #20: Auto Power on

blee1170, 2017-03-21:
Is there a way to have it automatically power on the pi when power is restored?

Basically, have everything powered on and running, lose power and it will continue to run off the battery. When that is depleted, it shutdown the pi. Once power comes back, the pi turns on and everything is happy.

NeonHorizon, 2017-03-21:
Not that I know of but if you find one let us know :)

blee1170, 2017-03-21:
Whelp. ¯\\_(ツ)_/¯

closed 2017-03-21 by blee1170

NeonHorizon, 2017-03-22:
You might be better off looking for a hat?

cgalpin, 2018-11-03:
I know this is closed, but it's been a while so I figured it's worth asking if anything has changed here? I hope it's not bad form to port to a closed issue.

Daniel, this looks like it's perfect for my project except for this one issue. Let me explain what I want to do in case I don't understand what LipoPi does. What I want to do is use a raspberry pi zero w like a usb dongle using a zero stem for example, where power is supplied to the pi directly from the device it's plugged into

https://zerostem.io

However the device it's plugged into is a car, and when the car turns off, I want to use a 1000c and LipoPi to allow the pi to run for a few minutes - long enough to connect to a wifi network, sync up some files, before shutting itself down. But then I want it to power up again when the usb port goes live (car turns on) as well.

Thoughts? If not able to be done with the 1000c and LipoPi, can you recommend a hat? The only thing I have seen so far that would work is the UPS PIco HV3.0B HAT, which is a bit pricey, and in the UK.

tia,
charles

cgalpin, 2018-11-03:
Yes, the idea is to plug it in and never touch it (until it needs a battery replacement). I could always ssh in and shut it down if needed while it's in the car with it on.

So I think your idea would make this work for me, thanks!

Whoohoo, I'm off to order some parts :)

NeonHorizon, 2018-11-03:
What I meant was if you did shut it down you wouldnt be able to start it again without killing the power.

cgalpin, 2018-11-03:
Sorry, I should have said that wasn't a problem afaikt. I'd just unplug and then plug it back in if needed right?

I'm ok with this sort of thing to do maintenance. I just want the general use case be that it happy powers up and down with the car, staying up just long enough to do a few things before shutting down cleanly.

And I think it's implied, but I shouldn't have any problem wiring up the power via the male usb port on the zero stem to the 1000c instead of via versa correct? If not, I know I can not connect power via the OTG port and get power via another usb port in the car to the 1000c but it would be cleaner to not have to have the additional wiring.

Thanks again

NeonHorizon, 2018-11-04:
Yeah you could unplug it or turn off the ignition which should cut the power.
The power would have to go into the 1000c only, you can't connect the power to the zero.