NeonHorizon/lipopi #21: Is it possible to use a ON/OFF switch with 2 positions?

propergol, 2017-05-13:
Hello,
first sorry if this is not the right place to ask you question. If this is not, then could you point me to the right forum?

I would like to know if your project could be moded easily to use an ON/OFF switch rather than a push button.
I am building a wifi radio for my mother and that would be much simpler for her to use a conventional ON/OFF.
thanks

NeonHorizon, 2017-05-13:
Yeah I've not tested but I think it would be possible....

You could try replacing the push button with a double throw switch. Literally just wire it in place with the batt line going to the switch common and the other pushbutton wire going to the switch contact for the on position. This would allow you to switch the system on in a similar way to when the pusbutton was pushed. Then you need to find a way to switch it off...

To do that connect the other terminal on the switch to a GPIO (Eg. GPIO 18) via two diodes and a 10k resistor. This will then pull GPIO 18 high when the switch is in the off position. You would then need to modify the shutdown script to shut down the Pi if GPIO18 goes high (as well as if GPIO 15 is low like it does now when the battery is flat). The diodes are to drop the battery line down to below 3v3 for the Pi as described in a similar circuit here: https://github.com/NeonHorizon/lipopi/blob/master/README.power_up_power_down.md

The only side effect I can see is if you shutdown the Pi with the switch left in the on position your battery will go flat as the switch will be holding the system powered up. But then because you are shutting down with the switch that shouldn't be an issue.

propergol, 2017-05-15:
many thanks for your help!
I have change my mind and for this radio project I will use a read only system so I can simply shut the PI down shorting EN and GRND on the Powerboost.
Anyway I will copy your project for other project ideas I have in stock :)
Thanks again 👍

closed 2017-05-15 by NeonHorizon