NeonHorizon/lipopi #23: raspi-config sets enable_uart=0

jeanrocco, 2017-06-09:
Hello, I don't know if I'm doing this right... 
The raspi-config on recent Raspian allows to disable the console with "A8 Serial", but it also sets enable_uart=0 in /boot/config.txt . Switching it back to enable_uart=1 makes GPIO_14 works as your documentation says. I'm on a rpi B+ V1.2. Great work thank's.

NeonHorizon, 2017-06-09:
If its working you are doing it right :)
Does it stay powered up when you start up and power off the Adafruit when you shut down?

jeanrocco, 2017-06-09:
by "...doing it right..." I meant I wasn't sure that rising an issue was the right way to contact you, it is obviously... I was just hinting that the documentation should reflect you have to adjust "enable_uart=1", with the present raspi-config... 
The GPIO_14 comes ON almost instantly when I apply power to the Pi. When I do a "sudo poweroff", GPIO_14 stays ON for about 5 seconds, turns off, but the activity green led on the Pi, will then blink rapidly 9 times and shuts off. 
If I understand your whole project correctly, this means it is safe to power down the Pi with GPIO_14 going down, even though the activity led is still blinking it's last 9 blinks. By safe I mean no SD corruption, this is the most important thing.

NeonHorizon, 2017-06-09:
Ahh LOL, yes this is the right way to contact.
One of the issues we appear to have is the different model Pi's seem to be different and also things seem to change with different versions of Raspbian and raspi-config. Basically we need someone with a few Pi's to check it out and let us know so we can update the documentation. Unfortunately I don't have a Pi3 so it won't be me :(

jeanrocco, 2017-06-09:
Hi, it's ok, I have a few Pi's, and telling you it does that :-) on my Pi2. I didn't try it on my Pi3 since it is not critical, I just use it to run GIMX. Can you answer the other question I had; no SD corruption even if we shutdown the power with GPIO_14 going down...and the activity led is still flashing.  This is important since I'm running "Home Assistant" on that Pi. It has to be reliable 7/24. Thank's !

NeonHorizon, 2017-06-09:
The Pi3 is very different as the UART is physcially connected to the bluetooth chip instead of the GPIO pins. The UART on the GPIO is emulated in software which is why it needs to be handled differently...

So just to check (so I can update the page https://github.com/NeonHorizon/lipopi ), on your Pi2 you ran raspi-config and selected no to "Would you like a login shell to be accessible over serial?" and yes to "Would you like the serial port hardware to be enabled?" and it set enable_uart=0 which stopped GPIO14 from going high is that correct? Or have the raspi-config options changed?

I can't answer your question 100% about the activity light but my understanding is those 9 blinks are a hardware process to tell you the system has shut down and are not related to activity with the SD card so you should be OK. At least nobody has reported any corruption and at that point Linux should be well and truly shut down.

jeanrocco, 2017-06-09:
Right answer: the raspi-config options changed... 

Pi2: Linux version 4.4.21+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #911 Thu Sep 15 14:17:52 BST 2016
Issue: Raspbian GNU/Linux 8 \n \l  (Jessie)

Pi3: Linux version 4.4.33-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #927 SMP Sat Nov 19 18:15:38 GMT 2016
Issue: Raspbian GNU/Linux 8 \n \l  (Jessie)

On both Pis, raspi-config, although slightly different, behave the same for Serial : 

sudo raspi-config
...Advanced Options
......A8 Serial
.........Would you like a login shell to be accessible over serial
...........NO

There is no option for the serial port hardware... this is why we have to edit /boot/config.txt and change enable_uart=1 at the end, since raspi-config sets it to enable_uart=0. 
Besides, on the new Raspbian Jessie release as of May 2016 serial port is disabled by default , it has to be enabled in /boot/config.txt!
" https://www.cube-controls.com/2015/11/02/disable-serial-port-terminal-output-on-raspbian/ "
I didn't run any tests on the Pi3 yet, only on Pi2.

Ok I guess, since there were no corruption reports, that it is safe to power down with GPIO_14 going down. Your answer might not be 100% but it is good enough and I thank you for it. This is great work since it corrects the major Pi's flaw, SD corruption.

NeonHorizon, 2017-06-24:
Sorry I didn't notice you responded.
I've updated the readme and added you on the contributors list. Could you check it's correct?
Thanks

jeanrocco, 2017-06-24:
Hello Daniel, this looks ok, considering most people will nowadays use Raspbian (Jessie). 

In order to really secure my Home Assistant server (rpi2) I finally ended up setting it up with your instructions and hardware (1000C), but for completeness added a small usb self powered hard disk drive, on which I mounted “ / " ( see https://github.com/adafruit/Adafruit-Pi-ExternalRoot-Helper <https://github.com/adafruit/Adafruit-Pi-ExternalRoot-Helper> ) and then set the SD to "ro" in initab. This way the SD is never written to but is only read from on boot followed by a system boot from the hard drive. I will soon replace the HDD with a small (silent and low power) SSD (20 gigs for 20$ can.) with this adapter: http://www.ebay.com/itm/252738735293 <http://www.ebay.com/itm/252738735293> . 

After much Googling I realize this combination was probably the most reliable way of setting up a server with the Rpi(s). Of course, this is all true only if the SSD is more reliable than the SD, which it is supposed to be. 

Thank you for adding me to the contribution list, I might have other ideas I could share… if they work :-) !

bye

jrb.

NeonHorizon, 2017-06-24:
SSD will definitely be more reliable than SD. Your only issue might be performance; the Broadcom CPU has a single USB2 port which goes to a hub built into the Pi itself which then supplies the built in USB ports and the network. This means if you use a USB based HDD you have to share a USB2 port bandwidth with the network which can be slower than using an SD. This may or may not be an issue for you.

closed 2017-06-24 by NeonHorizon