NeonHorizon/lipopi #25: Single spdt switch working

hex007, 2017-07-07:
I was wondering if something like this will probably work. The goal is to switch the EN to float for some time and then close it (just like Push button NC). This will cause the powerboost to stay on till Pi is booted and automatically switch off once Pi boots down. I tried making this and I faced two problems. 

- Without R1 the Pi causes speaker to emit high pitched noise.
- The Uart Tx doesn't go low but stays high after powering down hence power boost stays on.

```
   Uart Tx  o-----'\/\/\,------o,
                     R1          'o----------------o  EN
         Floating ------->     o  
```

NeonHorizon, 2017-07-07:
What version Pi is this?

hex007, 2017-07-07:
Pi1

NeonHorizon, 2017-07-07:
OK please disconnect your electronics, power up your Pi with a normal power adaptor and measure UART TX. Then shut it down and measure UART TX. What voltages do you get? You should get 3v3 when its on and 0v when its shut down (even with the power connected).

hex007, 2017-07-07:
I dont have a multimeter. Does it make a difference shutting down via `sudo halt` or `sudo shutdown` ?

NeonHorizon, 2017-07-07:
I wouldnt have thought it matters which you use. I don't know about the speaker noise I've not used LiPoPi with audio. Lets see what voltages you get as it may be that you have a wrong wire.

NeonHorizon, 2017-07-07:
If you dont have a multimeter use an LED with the right size resistor.

closed 2017-08-30 by NeonHorizon