NeonHorizon/lipopi #26: question of software install on/off version

jaysond04, 2017-08-01:
im new to linux and am having trouble setting up the service with systemd.

my project is a gameboy pi and have retropie in my pi right now and i have set up the python rpi.gpio 

i have downloaded the lipopi.py and lipopi.service  in my windows computer and using ssh with winSP software to transfer files to and from my pi. and now need to follow step 2. however theres a error saying permisson denied.

I also dont know what the follow means 
 Modify the file lipopi.service to point to where your copy of lipopi.py is located, then
$ sudo cp lipopi.service /etc/systemd/system/.

can someone direct me to what i need to do or have directions thats some one thats starting can understand :) .

thank you

NeonHorizon, 2017-08-01:
To do the copy you need to use sudo, thats probably why you are getting permission denied.
To modify the file edit it in a text editor and change the patch to wherever you plan to keep lipopi.py

jaysond04, 2017-08-02:
thanks for the replay, 

your comment help me copy files around but i would like to clarify something though, on step 2 you say " copy service file to /etc/systemd" but then later on, in your script in  step 2  it looks like your copying the  lipopi.service file to directory etc/systemd/system/   

are  those 2 different times you copy the service file, one in /etc/systemd and another copy to /etc/systemd/system?

i went ahead and assumed you meant you wanted the lipoi.service file modified and then just placed in /etc/systmd/system. i then went ahead and followed step 3. 

i thought i was finished but hen i press the switch to turn it on, the pi and the power boost only turns on when im pressing and holding the push button. and when i let go it turns off. what am i doing wrong?
the type of switch im using is this

https://www.banggood.com/50PCS-TC-1212T-12x12x7_3mm-Tact-Tactile-Push-Button-Momentary-SMD-PCB-Switch-p-1000837.html
[image]

and my wiring diagram for powerboost and pi zero is the above.  i just used one leg of the switch to connect the diode portions:

NeonHorizon, 2017-08-02:
Sounds like you've not configured the UART correctly in Raspiconfig. Disconnect your circuit, disable the systemd service, power up the Pi in the usual way using a USB adapter and check GPIO14 is high.

jaysond04, 2017-08-03:
Haven't had time to disconnect the circuit and disable the systems service  to check if gpio 14 is high but did have time to check the config.txt file if that's what you mean by raspiconfig.  In the last line it alrrady says enable_uart=1 (as shown in picture). Is that what you mean by uart not correctly configured since in the basic lipopi instructions you mentioned about changing uart=0 to uart=1

NeonHorizon, 2017-08-03:
You need to test it really. We've had lots of different issues with different Pi's and different versions of Raspbian.

jaysond04, 2017-08-04:
ok i disabled the systemd service by typing:

 $ sudo systemctl disable lipopi.service

and was able to disconnect everyhing from the pi zero w and powered it through the micro usb power port.
Using the multi meter i i check the voltage on gpio 14 and it read 3.11 volts.

NeonHorizon, 2017-08-04:
Excellent, and when you shut down it goes to zero even if the power is still connected right?
If that is true and it still powers off when you let go of the button (and the service is disabled) you must have your circuit wrong or your LiPo is flat.

closed 2017-08-30 by NeonHorizon