NeonHorizon/lipopi #27: LiPoPi Power Up / Power Down version low power reset

robinsdl, 2017-08-30:
I would like to add a low battery warning reset feature to your code. I have a project that is solar powered and I would like it to be able to reset the low battery shutdown if the battery starts to be charged by the solar panel. I have tried myself but can not get it to work. Please help.

NeonHorizon, 2017-08-30:
Surely if the battery is being charged it will no longer be a low battery? I don't think its a good idea to disable the low battery shutdown as you may drain the battery too low and end up with the Pi being killed as the PowerBoost shuts off?

Anyway to do what you are asking I would guess you would want to use a different GPIO pin as an input which your solar charging circuit would toggle high or low when its charging. Then you need to change the lipopi.py script to check this and disable (or at least postpone) shutdown.

robinsdl, 2017-08-30:
What I would like the script to do is to stop the low battery shutdown if the low battery input changes state e.g. Is the battery is no longer low. To do this the script needs to monitor the low bat GPIO input and stop/reset the low battery shutdown. At the moment the script stops looking for inputs and waits the shutdown. How do I alter the script to keep looking for input changes while waiting to shutdown?

NeonHorizon, 2017-08-30:
OK if you check lipopi.py theres a function called lipopi_low_battery_shutdown(). What I would do is after the line time.sleep(lipopi['shutdown_wait']) add a couple of lines of code to check if the lipopi['low_battery_pin'] is high and if it is return from the function instead of continuing to the message and shutdown command. That should do what you want. Backup your SD card before testing though just in case it doesn't :)

closed 2017-08-30 by NeonHorizon

robinsdl, 2017-08-30:
I have tried many different versions of having the code check to see if the pin goes high again without success. I must be doing something wrong. Could you please tell me what code to use.

NeonHorizon, 2017-08-30:
I'm not really a python guy but this looks really simple...
Have you tried:
```
if(GPIO.input(lipopi['low_battery_pin']))
    return
```

robinsdl, 2017-08-31:
Yes I have tried that and it doesn't work. I don't have a lot of python experience either which is why I was asking. I will just use your original script and and add in an email notification when the battery goes low.
Thanks for the help.

NeonHorizon, 2017-08-31:
What do you mean by it doesnt work. Did you get an error or anything?

robinsdl, 2017-08-31:
When the code you suggested is added the script stopped working all together. I did not see any error codes however I just realised that I may not have seen them as I rebooted and then tried the script. I will need to test it again by stopping and starting the script. The unit is currently deployed to test the solar charger so it will be a few days before I can test it.

NeonHorizon, 2017-09-01:
Its probably worth running the script manually from the command line to test rather than from the systemd service.