NeonHorizon/lipopi #28: RPI3 with Stretch works like the others

jeanrocco, 2017-12-01:
Hello Daniel
The RPI3 works fine with Rasbian Stretch, it behaves like the othe ones. 
I have been inspired by you lipopi project and came out with https://github.com/jeanrocco/Pi_SD_SAVER . If you have some time to spare, would you  look at it and tell me what you think... 
jrb.

NeonHorizon, 2017-12-01:
Nice project :)
DOes that mean I should remove the special notes about the RPI3?

jeanrocco, 2017-12-01:
Hello Daniel,

> 
> Nice project :)
Will you try it ?  :-) ! it’s an eyes opener for the inexpensive Attiny85
> DOes that mean I should remove the special notes about the RPI3?
I didn’t check with priors Raspbians, but I think you note is still valid for the earlier versions. It will be up to the readers/users to find out what they should do, there are too many variations already. But with your project they are forewarned of the potential problem.

I just got this brand new RPI3 from Arrow USA, a black Friday special, only 45 $ Can. flat, no taxes no custom fees, a bargain considering it sells here in Montreal close to 80 $ with taxes. I downloaded Raspbian Stretch and was please to see it was handling the TX signal like the previous PIs’s Os.

I then simply swapped this new RPI3 (stretch) with my RPI 2+ (Jessie) that I was using for development and my PI_SD_SAVER gadget worked perfectly.

NeonHorizon, 2017-12-02:
Thanks for the info I will update the page to reflect this.
Unfortunately I have no time for projects at the moment as my work is very busy so I wont be able to check your ATTiny project :(

closed 2017-12-02 by NeonHorizon