NeonHorizon/lipopi #29: gpio switch from pitft gpio

darkmoon3d, 2018-02-07:
Am wanting to use a gpio pin specifically pin 17 on adafruit pitft 2.2" (https://www.adafruit.com/product/2315) . Is this possible? Am also looking into a failsafe switch where user would turn the device over while holding down the button in order to turn the device on or off. Would use tilt switch to detect when device is turned over (https://www.adafruit.com/product/173).

NeonHorizon, 2018-02-07:
I'm not sure I understand your question but the basic LiPoPi project only uses GPIO 14 and 15 so it won't conflict with the Pi TFT. The Power Up Power Down version uses GPIO18 which will conflict so you will need to change this to a different pin (change shutdown_pin in lipopi.py).

For the tilt switch you would probably need to put this on its own GPIO pin and modify the code in lipopi.py to detect the position of the switch and the power button hold before shutting down.

closed 2018-02-07 by NeonHorizon

darkmoon3d, 2018-03-09:
This is more along lines of what trying to build. Have managed to test out circuit on single momentary push button but unable to use pitft 17 button. Unsure if battery is too high of voltage and requires capactiors or diodes or is electrical flow is conflicting. Code is working where device is shutdown when pressing 17 and correctly shuts off but startup is issue.

https://iu.box.com/s/iouw9qqhhemwl03w3ii8o382bo2vdrou

NeonHorizon, 2018-03-09:
OK I think I see the problem...
The button you are using has one side connected to GPIO 17 and the other side connected to ground. This wont work as the button needs to have the other side connected to bat and the other connected to the diodes. You wont be able to resolve this without doing a lot of modifications to the PiTFT and its likely to not even be possible depending on how they have routed the tracks on the PCB. You would basically need to cut the track that joins the button to 17 and cut all around the earth going to the other side of the button, then solder wires directly to the button going to bat and the diodes. This is likely a bad idea.