NeonHorizon/lipopi #32: GPIO 14 does not stay high on power up (Pi Zero W version 1.1)

svenyonson, 2018-04-18:
I have disabled the serial port, although the raspi-config menu has changed for Raspberry Pi W version 1.1.
Instead of Advanced, you have to use Interfacing Options->Serial. 

Testing by  leaving EN unconnected and inserting the battery to turn on the Pi W, and with a voltmeter on GPIO 14 ( pin 8), the pin goes to 3.3v as soon as it powers on, but then goes low again after a second or two. So, when the power up/down circuit is implemented as described in the README, it does not latch in the power up state. I'm looking at the meter now, after the Pi has been running for about 5 minutes, and the voltage is .75 volts.

svenyonson, 2018-04-18:
Raspbian version 9 (stretch)

svenyonson, 2018-04-18:
Ok, testing now, but it appears that UART RX stays high when powered up. I will reimplement the power up/down circuit using GPIO 15 and see if it works. The pin is constant at 2.47v which seems enough to keep EN happy.

NeonHorizon, 2018-04-18:
Does RX go low again after shutdown though?
If it doesn't the circuit wont work.

Unfortunately they keep messing with the serial port settings and its almost impossible to keep up. I would keep trying different combinations until you find one that makes TX go high. Did you try the enable_uart boot setting?

svenyonson, 2018-04-18:
Hmmm, nope it doesn't, so it doesn't work. I don't know about the enable_uart setting, I'll have a look.

svenyonson, 2018-04-19:
Great, thanks Daniel. enable_uart fixed the problem. Might want to add that to the README.

Thanks again,

Steve

NeonHorizon, 2018-04-19:
It is in the README?
Step2 in the software setup.

closed 2018-04-19 by NeonHorizon

NeonHorizon, 2018-04-19:
Ahhh sorry, I didn't realise you were looking at the power up/down version, thanks for the tipoff I've updated it.

I had a look at your circuit, I have to say I'm not great at RC networks either and there's a lot going on in that circuit so I couldn't say for sure if it would work but I can't see any harm that could come from trying? 

The only possible issue I could see is theoretically there could be infinite current from bat through push button and diode to the capacitor during initial charging of the 4.7u. Its a small enough cap that it likely wont matter though but if you wanted to be really cautious you could put a small resistor, say 100 ohm between bat and the pushbutton just to keep it under control.

svenyonson, 2018-04-21:
Well, worth a try, but no luck. The 2M (actually 1.8M) is to much resistance to be an effective pull down resistor, and so EN never goes low. Somehow I need a way to isolate the RC network from EN such that it holds it high for the required period without letting the cap prematurely discharge through the 100K pulldown.

svenyonson, 2018-04-21:
Now I am heading into dangerous territory, as I only have a slight clue as to what I am doing.  
[image]

NeonHorizon, 2018-04-22:
Transistor is the wrong way around I think, should be pointing downwards.

svenyonson, 2018-04-22:
Whoops, I was disoriented. Yes, upside down. I have the transistors on order, I'll try it out then.

svenyonson, 2018-04-22:
Another suggestion I received on electronics.stackexchange does the EN pulldown through the UART. Not sure if this would work, but interesting idea:
[image]

NeonHorizon, 2018-04-22:
Yes that may well work.

svenyonson, 2018-04-22:
No, don't think so. Powered off, there is 29M resistance between UART Tx and Gnd.

NeonHorizon, 2018-04-22:
Yeah that sounds to high for the pulldown to be active, the integrated pull up/down resistors are like 50k I think.

svenyonson, 2018-04-24:
I have a working solution. Very similar to your original, with the addition of a capacitor and I replaced the two diode voltage step down with a voltage divider - which actually reduced the component count because the 22K also serves the same purpose of the original 10K on GPIO 18.

So, press/release to power on. Capacitor keeps EN high for a bit more than 10 seconds so UART TX can stabilize. Press again to power off. After the Pi shuts down, EN keeps the powerboost on for an additional 8 seconds or so, but not an issue.

[image]

NeonHorizon, 2018-04-24:
Awesome!
Would you be prepared to do a pull request to include it in the project?

svenyonson, 2018-04-24:
Sure!

svenyonson, 2018-04-24:
I have a branch to push for a pull request, but I don't have write access

NeonHorizon, 2018-04-25:
Yes thats correct, I can't remember how you do it exactly but I think you commit it to your own repo then make a pull request or something?

svenyonson, 2018-04-25:
The way I've always seen it done is that contributors create a branch (never commit to master) then push the branch, create a pull request for you to approve and merge into your branch. But that's always been in the context of a single organization, I haven't committed to someone elses repo or open source repo's before. 

If I were just to commit to my own repo (which I already have) then you wouldn't ever be able to see the changes (on my machine). Somehow my branch needs to be pushed to a place where you can see it (and the diff with your master).

If it is easier, I can email you the image and updated readme and you can merge and commit yourself?

NeonHorizon, 2018-04-25:
Does this help?
https://help.github.com/articles/creating-a-pull-request/

svenyonson, 2018-04-25:
Brilliant. I've learned something new!

You should see the pull request now.

svenyonson, 2018-04-25:
P.S. The "./" path changes in the readme are so you can see live preview of the markdown within your local editor. It shouldn't affect how it is displayed in GitHub. (I am using Sublime Text 3 for my editor, and the markdown plugin is MarkdownPreviewLive which gives you a nice side by side view.

NeonHorizon, 2018-04-27:
Thanks for that!
Sorry it took so long to reply I've had a crazy week.
I'll also add you to the contributors list.