NeonHorizon/lipopi #34: Pi Zero W powering as soon as battery is connected

Goeste, 2018-05-01:
Hi,

I have an issue with the setup:

* I am using (I want to use) the power up/down version
* Everything is connected as it is shown in the images (Pi Zero W & 1000C).
* Unfortunately every time I connect the battery the pi starts without interaction.
* The script is running fine.
* Serial setup via raspi-config is "no" and "yes"
* enable_uart=1 is present in config.txt
* shutdown via button press works like a charm
* turning on the pi via button press is NOT working, independent on how long I keep the button pressed
* disconnecting and connecting the battery again powers up the pi...
* power is soldered directly from 1000C to pi (no usage of USB)
* Distribution is Raspbian Stretch (up2date)

Can anyone advise where to look for an error?

If I need to provide more infos please let me know.

Best Regads,
goeste

NeonHorizon, 2018-05-01:
It looks like this is an electrical problem not software.
Please check the 10k, 100k, 1N4001 and Enable are all connected, this is not a crossing this is a join. Without a battery connected, measure with a meter and make sure you have 100k resistance between ground and enable on the 1000C.

NeonHorizon, 2018-05-02:
It looks right from the pictures, you are missing a photo of the 1000C.
Did you try a test meter between EN and Ground on the 1000C to measure the resistance?

Goeste, 2018-05-02:
[image]

unfortunately i cannot measure it... but with 2 setups both are not working correctly...

NeonHorizon, 2018-05-02:
The Bat wire on the 1000C is yellow but its blue on the breadboard?
Other than that it looks OK from what I can see.
What is the voltage on G4 on the breadboard? It should be 0v.

Goeste, 2018-05-02:
yh, true, yellow=blue since I didn't happen to have more blue wires... have to measure late today... though I guess there is a voltage since the pi starts as soon as the battery gets connected... will measure that if somewhere is a shortage...

Goeste, 2018-05-08:
ok I checked it. G4 is at 3,7V (constantly after connecting the liion-pack BUT pressing the button brings it down to 0V... somehow an upside-down world...

however... when the pi is running and i press the button, the pi shuts down as it is supposed to be. but it cannot be started... afterwords

NeonHorizon, 2018-05-08:
I think if I had your circuit on the bench in front of me I might be able to help but its very difficult in here. I would recommend you contact a local maker/hackspace for help?

closed 2018-05-08 by NeonHorizon