NeonHorizon/lipopi #41: EN pin on Vs instead of Bat ?

rtatibouet, 2018-11-19:
Hello, I was wondering if there was a specific reason you user the Bat pin to activate the EN pin, and not the Vs pin ?

With the current setup, when no battery is present, the button does not work. Seems legit, no battery=no Bat output

My idea is to use the Vs pin to activate the EN pin instead of the Bat pin, so I can use the button even when only the USB cable is plugged in. It can go as high as the USB voltage, but the diodes should help protecting the GPIO anyway.
Am I wrong somewhere ?

Thank you for your reply.
Ronan

NeonHorizon, 2018-11-19:
Hi Ronan,
Yes that should work although like you say the voltages would be a bit too high. It would stick 5v - 0.7v -0.7v = 3.6v into GPIO 18 which is likely fine and 5v - 0.7v = 4.3v into GPIO 14 which could be an issue.

It might be worth moving one of the diodes which is inline with GPIO 18 so its inline with Vs and the button so it drops both GPIO 14 and GPIO 18.

closed 2018-11-19 by NeonHorizon

rtatibouet, 2018-11-19:
I just figured this:
https://forums.adafruit.com/viewtopic.php?f=8&t=143213&p=709615&hilit=powerboost#p709615
Adafruit guy says we shouldn't user the 1000C without a battery plugged in at first.

I did this a few times... Hope I didn't killed my charging module. It has a weird behavior

So the use-case for my setup just disappeared, as the 1000C is not intended to work that way.

Thanks for the reply.

NeonHorizon, 2018-11-19:
Thats interesting, I didn't know that thanks!