NeonHorizon/lipopi #44: Recommended code change

RChadwick7, 2019-02-04:
While everything works fine on the Pi Zero W that has the shutdown circuit on it, while testing on an unmodified Zero W it immediately shuts down. Looking at the code, it appears it sets a pulldown on GPIO18, but it appears a pullup is needed at GPIO15. Is there a reason this isn't a good idea?

    # setup the low battery check pin
    GPIO.setup(lipopi['low_battery_pin'], GPIO.IN)
changed to:
    # setup the low battery check pin
    GPIO.setup(lipopi['low_battery_pin'], GPIO.IN, pull_up_down=GPIO.PUD_UP)

NeonHorizon, 2019-02-05:
I guess it would depend on if the low battery output on the PowerBoost goes low or if it goes open circuit when the battery is low. Have you tested it to see if it still shuts down on low battery?

RChadwick7, 2019-02-11:
I've tested it, seems to work fine, at least for me. I haven't had a chance to check the datasheet of the chip used on the Adafruit inverter, but I would assume it would give a good solid logic high and low.

NeonHorizon, 2019-02-11:
Excellent, would you like to submit a pull request?

RChadwick7, 2019-02-11:
I've never done a pull request, nor am I sure what it is.  Any advice?

NeonHorizon, 2019-02-11:
Its how you contribute to open source code on Github :)
Basically you request a change (pull request) and the maintainers accept it and it gets merged into the project.

I can understand if you'd rather not but its quite rewarding if you did want to get involved, once you have helped one project it can get addictive and you end up contributing quite a bit even if its just correcting spelling mistakes in various projects :)

RChadwick7, 2019-02-11:
I'm very happy to contribute if I can, but didn't know how, and not everyone is interested. I'll see what I can figure out :)
Also, I learned enough Python to add a function where the Pi won't shut down unless if you've held down the power button longer than a pre-defined  amount of time. I cloned your Github  and made the change there. Should I do a pull request for that as well? I assumed not everyone would be interested in that, although in my case, I'm using a button in an existing item, and wanted to share functionality.

NeonHorizon, 2019-02-11:
I think people would be interested in it but it would be best if they can enable the functionality optionally? Perhaps by setting a variable to turn it on?