NeonHorizon/lipopi #46: step 13 not working. Pi powers off when releasing button

MatayMayrany, 2019-05-13:
I'm trying to configure the simple set up that powers up after a few second of holding the button but the power boost and the pi both shutoff once i release the button... 
https://imgur.com/a/k45g19E
this is my set up so far. 
help pls

NeonHorizon, 2019-05-13:
Please make sure your battery is charged, also you are missing the 100k resistor to ground.

Also I think this is a PiZero-W is that correct?
If so you may need to do step 1 of the software first.

MatayMayrany, 2019-05-13:
Ok I do have a resistor on the backside Of the boost. It is a pi zero W but I hadn’t done the first step in the software yet. Will try, thanks for the fast reply :)

MatayMayrany, 2019-05-13:
it worked with the step one in the software!

closed 2019-05-13 by MatayMayrany

NeonHorizon, 2019-05-13:
Great news!
I'll update the instructions as a reminder.