NeonHorizon/lipopi #54: Variant for reusing switch for soft powerdown

dtourolle, 2020-08-06:
Hey Guys,

Really nice project and guide. I made a simple adaptation of your work which allows the same switch to also be used for a soft power down. It needs a few extra resistors and two transistors, the "battery button" circuit now activates two transistors, one enables the powerboost while the other pulls a GPIO to high. This can then be soft captured as a shutdown signal. Writing this I realise a diode would also work in place of the first transistor, so the second circuit is an untested alternative.

[image]

[image]

I have used the first circuit and could power on and off the RPI using the same switch. If you want to include this as an alternative config in your guide I have tried to create these figures as close possible to your existing ones.

Cheer,
Duncan

dtourolle, 2020-08-06:
I think this also solves #45?

NeonHorizon, 2020-08-08:
Hi Duncan,
Looks good, although I think we already have this covered on this page?
https://github.com/NeonHorizon/lipopi/blob/master/README.power_up_power_down.md

It seems the only difference is you are using a transistor instead of two diodes?

dtourolle, 2020-08-11:
Sorry, I didn't see that page. I guess the only advantage of my setup would be if the battery were to drop bellow 3.3V as then then powerdown gpio would not be triggered for the diode version?

closed 2020-08-11 by dtourolle

NeonHorizon, 2020-08-11:
Your version definitely seems to have better 3v3 signal handling. When I get a moment I will add it to the page.

NeonHorizon, 2020-08-15:
Duncan I've added your circuit to the power up/down version page and added your name as a credit on the main page, hope thats OK :)  (also I hope I guessed your name right LOL!)

OopsDoh, 2020-10-17:
@dtourolle Did you make any changes in the [lipopi.py](https://github.com/NeonHorizon/lipopi/blob/master/lipopi.py) file to get your version to work?

In your diagram you are connecting RaspberryPi's 3V3-GPIO with an 10K resistor and the transistor to "any GPIO" - in my test I used GPIO 23 and replaced 18 by 23 [in this line](https://github.com/NeonHorizon/lipopi/blob/914570c9a340abe9abed0ef63755393367e1acd2/lipopi.py#L98):
`lipopi['shutdown_pin']    = 23`

In the script the shutdown GPIO is set as an input with a activated pulldown resistor ([line 31](https://github.com/NeonHorizon/lipopi/blob/914570c9a340abe9abed0ef63755393367e1acd2/lipopi.py#L31)):
`GPIO.setup(lipopi['shutdown_pin'], GPIO.IN, pull_up_down=GPIO.PUD_DOWN)`
Should this be changed when using your circuit?

What type of diode did you use in your second version? Is a "1N4001" fitting in this circuit?

I tried both of your circuits/diagrams but both versions didn't work. The red LowBat LED of the PowerBoost is starting to light up as soon as I connect the battery and the RaspberryPi doesn't even boot. I checked the wiring multiple times, checked my battery (is full and working with the other version of NeonHorizon), replaced the transistors and even replaced the PowerBoost 1000C as well as the RaspberryPi (Zero W in my case) with new devices - but it still didn't work... I don't know what else could be wrong beside missing changes in the lipopi.py file...

@NeonHorizon Why do you think that this version does have a better 3v3 signal handling? The transistors and 3v3 connection is only relevant in a fraction of a second when the button is pressed - or did I miss something here?

I think a "better 3v3 handling" would be required for the direct connection of GPIO 15 (UART RX) to the LB pin (Low Battery) of the PowerBoost, because the signal of the LB pin can flicker when there are short peaks in current. For example when using RGB LED stripes that are powered directly by the 5V output of the PowerBoost and getting the RGB signals by the Raspberry (also powered by the 5V output of the PowerBoost). In my test with [your "power up - power down" version](https://github.com/NeonHorizon/lipopi/blob/master/README.power_up_power_down.md) (I used the second version with capacitator) the Raspberry is shutting down quite often automatically because of short peaks in current even when the battery is charged full.

Do you see a way to include some kind of "buffer" in the connection of GPIO 15 to the LowBat pin to avoid this kind of shutdowns because of short peaks in current? I thought about another capacitator but I'm no expert in electronics and don't know if this is a good idea or how the circuit for this connection should look like. Maybe this isn't necessary at all because there is a way to include a signal buffer in the python script? Eg some kind of delay to wait a second after the first LowBat signal and then check again if it is still low and then shut down?

dtourolle, 2020-10-19:
Hi OopsDoh, I only used the circuit as a basis for my own project. I did not use or modify any of the code.

OopsDoh, 2021-05-10:
Unfortunately no, I couldn't find a solution to get this version to work properly. I kept the previous version [A Simple Press (no hold) Power Up Version](https://github.com/NeonHorizon/lipopi/blob/master/README.power_up_power_down.md#a-simple-press-no-hold-power-up-version) and I'm still having trouble sometimes, because LowBat is triggering/flickering sometimes and shutting down to quickly.