NeonHorizon/lipopi #55: Power down while charging => immediately off

Felix79a, 2020-09-01:
I use the "Simple Press (no hold) Power Up Version", with one 1N4001 and Power Down connected to GPIO 6.
RPi 3B+, Powerboost 1000c, original charger.

Works fine, but when charging, it seems as if the button would create a short curcuit. There is no graceful shutdown - Pi goes immediately off. Any ideas?

NeonHorizon, 2020-09-01:
My guess is there's a fault with your circuit somewhere.
Check the capacitor isn't shorted for example.

Felix79a, 2020-09-01:
Thanks for your fast answer - will check it again.
But everything works as expected when no PSU is connected.

Powering up is also a bit confusing. Pressing the button with PSU - nothing happens. But when i remove the PSU, the Pi starts booting.

Felix79a, 2020-09-01:
OK, checked everything again. No fault on my pcb.

But i found out that it makes a difference using a non-insulated vs. insulated screwdriver to shorten my button-pins... This curcuit  (or GPIO) seems to be pretty sensitive? I wired a real switch now for testing.

Powering down now works as expected. But powering up with connected PSU works only randomly.
There is another curcuit printed here by Duncan. Is it more robust and can it be used with 1000c as well?

NeonHorizon, 2020-09-01:
Yes GPIO is very sensitive.
I've not tested Duncans circuity personally but it looks like it should work.

Felix79a, 2020-09-02:
why is no capacitor required in his version?
I thought it takes some time for TX/14 to come up?

NeonHorizon, 2020-09-02:
I think the capacitor was to compensate for data on the pin.

closed 2020-09-02 by Felix79a