NeonHorizon/lipopi #56: On-Off switch suggestion

miki-mulet, 2021-02-19:
I'm not very savvy on electronics, so I wanted to ask if this setup would make sense before I try it and burn something. I'd like to use a two position switch for both booting up my Pi Zero and shutting it down, changing the following from your 'Basic version':

[image]

Where the X pin is pulled high when the switch is on the OFF position, triggering a shutdown, and otherwise always on the ON position when the rpi is running.
Maybe it's total nonsense for obvious reasons or maybe you've already tackled this solution, in any case it'd be  very useful to know! Thanks in advance

NeonHorizon, 2021-02-20:
Its been a while since I looked at LiPoPi but it looks like the main issue with this is the switch in the on position overrides GPIO14 so if you shut down the Pi the circuit wouldn't disconnect the power unless the switch is switched off as well.

What you might be better off with is a 3 position toggle switch where the actions are off - run - start where the start position is momentary but the off is latching. So basically you switch from the bottom off position to the middle run position. Then momentarily push it up to the start position and release which causes the Pi to start. The switch has then dropped back to the run position which has nothing connected and the Pi can then power the circuit off if it needs to.

So basically the momentary start position is actually the same as the pushbutton in the original circuit and you have simply added an off position which as you say hopefully makes GPIO X go high.

I know this switch sounds exotic but they are actually pretty easy to get.

miki-mulet, 2021-02-20:
Thank you for the fast response! It seems that indeed I wasn't taking into account the possibility of an automatic shutdown not triggered from the on-off switch. 
The 3p toggle switch sounds like a good alternative, otherwise I might end up just using two separate buttons for on and off, for simplicity.

closed 2021-02-20 by miki-mulet