NeonHorizon/lipopi #58: Hum noise

stahlhelm98, 2022-05-21:
Hello! Using PowerBoost 500 while powering raspberry pi there is a hum noise and after few seconds the raspberry is shutting down. When i power raspberry pi from another source everything is ok. Do you know what the problem should be ? Thank you very much! I use a LiPo cell 2600 mAh and a USB cable.

NeonHorizon, 2022-05-21:
Difficult to say really.
Do you have a photo?
Also which Raspberry Pi is it?

stahlhelm98, 2022-05-21:
Hello! Thank you very much for the reply! It is a Raspberry Pi Zero.
[image]

NeonHorizon, 2022-05-22:
Is the battery fully charged?
From what you describe it sounds like the battery cant provide enough power but it looks adequate.

stahlhelm98, 2022-05-22:
> Is the battery fully charged? From what you describe it sounds like the battery cant provide enough power but it looks adequate.

The Li-Po battery is now 4.18V. It`s weird because if power a Raspberry Pi 4 there is no problem.

NeonHorizon, 2022-05-23:
Very strange, I wonder if the load is too small?
Have you tried connecting an LED on the PiZero or something like that?